NickStallman/home-assistant-repo #1: No Connection

yelhsa083, 2024-05-05:
Attempt to connect to System, logs indicate an error received, with nothing further.
Issue possible due to firmware in winet device
It also broke SunGather. 

Connects to MQTT broker

**Log Extract from Add On**
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started

> cli
> node build/src/index.js

Connected to MQTT broker
Fetched l18n properties:
2024-05-05 05:26:39 info: Connected to Winet
2024-05-05 05:26:39 info: Detected device: SH10RS (A########0)
2024-05-05 05:26:39 info: Detected device: SBR192 (S########5)
2024-05-05 05:26:40 error: Received message:
2024-05-05 05:26:40 error: Invalid realtime message:
{ service: 'real' }
2024-05-05 05:26:40 warn: Reconnecting to Winet
2024-05-05 05:26:45 info: Connected to Winet
2024-05-05 05:26:45 error: Received message:

**Log from Supervisor**
2024-05-05 15:25:27.860 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on b######5/amd64-addon-winet-extractor with version 0.1.0
2024-05-05 15:26:24.376 INFO (SyncWorker_2) [supervisor.docker.manager] Stopping addon_b######5_winet-extractor application
2024-05-05 15:26:27.698 INFO (SyncWorker_2) [supervisor.docker.manager] Cleaning addon_b######5_winet-extractor application
2024-05-05 15:26:37.999 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on b######5/amd64-addon-winet-extractor with version 0.1.0

**Log From Home Assistant Core**
Logger: homeassistant.components.hassio
Source: components/hassio/websocket_api.py:135
integration: Home Assistant Supervisor (documentation, issues)
First occurred: 3:00:58 PM (4 occurrences)
Last logged: 3:07:25 PM

Failed to to call /addons/b######5_winet-extractor/start - Unknown error, see supervisor
Failed to to call /addons/b######5_winet-extractor/uninstall - Addon is not installed
Failed to to call /ingress/validate_session -

**HA Version**
Core 2024.5.0
Supervisor 2024.04.4
OS 12.2
Front End 20240501.0

**Comms Module**
Firmware: WINET-SV300.001.03.P010

yelhsa083, 2024-05-05:
Should add, whilst it says connect to MQTT, there has been no device activated within the MQTT server

NickStallman, 2024-05-05:
Thanks for checking it out, have you changed the default password by any chance.

I've updated it (Addons -> Addon Store -> Three dots on the top right -> Check for Updates) to improve logging, and to allow specifying a non-default username/password if needed.

Just the addon logs should be needed if there are still any issues.

yelhsa083, 2024-05-05:
Updated
User & pw are still the default out of box, but updated in config

No device within MQTT
But seems to auth correct to winet 

Error log 

> cli
> node build/src/index.js

2024-05-05 09:20:34 info: Connected to MQTT broker 
2024-05-05 09:20:34 info: Fetched l18n properties. 
2024-05-05 09:20:34 info: Connected to Winet, logging in 
2024-05-05 09:20:34 info: Detected device: SH10RS (A########0) 
2024-05-05 09:20:34 info: Detected device: SBR192 (S########5) 
2024-05-05 09:20:34 info: Authenticated successfully 
2024-05-05 09:20:34 error: Invalid realtime message: {
  data: {
    result_code: 106,
    result_msg: 'I18N_COMMON_ACCOUNT_OUT_FRESH',
    result_data: { service: 'real' }
  },
  [Symbol(level)]: 'error',
  [Symbol(splat)]: [ { data: [Object] } ]
}
2024-05-05 09:20:34 warn: Reconnecting to Winet 
2024-05-05 09:20:39 info: Connected to Winet, logging in 
2024-05-05 09:20:39 info: Authenticated successfully 
2024-05-05 09:20:39 error: Invalid realtime message: {
  data: {
    result_code: 106,
    result_msg: 'I18N_COMMON_ACCOUNT_OUT_FRESH',
    result_data: { service: 'real' }
  },
  [Symbol(level)]: 'error',
  [Symbol(splat)]: [ { data: [Object] } ]
}
2024-05-05 09:20:39 warn: Reconnecting to Winet 
2024-05-05 09:20:44 info: Connected to Winet, logging in 
2024-05-05 09:20:45 info: Authenticated successfully

yelhsa083, 2024-05-05:
Not sure if it makes a difference to this implementation, when logging directly to the webpage of winet with admin, there is several pop-up windows that need to be closed prior
Password change prompt, setup system and another prompt. Once closed can then see and then go between details

NickStallman, 2024-05-06:
Working with another user, it looks like older firmware versions and the latest one handle authentication differently (older ones didn't enforce it properly at all).

I've made a tweak in 0.1.2 which should support the latest firmware version perfectly without breaking older firmware support. :)
Run a update and see if this does the trick.

yelhsa083, 2024-05-07:
Seems to connect and stay active.
Been connected/active overnight.

Have this integration, mkaiser modbus and isolar all active and all have stayed connected overnight, not causing lockup of the winet.
Had also setup within the winet device a white list for modbus, don't know if that has helped things too.

Believe that the mod on 0.1.2 has worked to allow connectivity.

NickStallman, 2024-05-08:
I'll add some handling and auto-restart for I18N_COMMON_INTER_ABNORMAL.

That translates to "Internal Error" which is interesting. :)
The code just needs to handle it better. That'll be in the next version.

NickStallman, 2024-05-08:
Version 0.1.3 is out which fixes I18N_COMMON_INTER_ABNORMAL.

yelhsa083, 2024-05-10:
Dongle crashed overnight, a separate integration autostarted (now stopped).
Been running this and mkaiser together since around 0700 this morning (after dongle reset) so about 12hrs. Going ok so far. 0.1.3
Need to set a automation notification from ping that dongle not responding to go and reset it.

The auto recover in 0.1.3 seems to be working, but seems to trigger quite regularly, may be is associated with isolar polling?

May look to reduce the 10s polling if possible to 15 or 30s reduce likely hood of disconnect once can get over 24hr with both active.
Is there an easily accessible variable?

NickStallman, 2024-05-12:
The new version is out with a poll interval parameter.

I'm also using mkaiser (with modbus-proxy), but I've removed all the duplicate sensors so it's fetching substantially less data and mostly only the settings like min soc, min/max charge discharge which don't seem to be available via Websocket.
This seems to be quite stable for me, especially with the recent error handling improvements.

yelhsa083, 2024-05-14:
Have so far been stable on 1.3
Just now have updated to 1.4, see how things progress
Thank you for your work

yelhsa083, 2024-05-21:
Think this issue can be closed, been stable, ie not crashed / hung. 
I've not needed to intervene with it at all to keep it operating.
Thank you for your work.