NickStallman/home-assistant-repo #2: Mqtt address

SNAPE15, 2024-05-05:
Thanks for the addon, hopefully its a sucess. I'm having an issue with connection to mqtt (mosquitto). I see the address format your using, but as far as using mqtt addressing before Ive only used core-mosquitto. Only used mqtt for GoSungrow before so not experienced in it. Do I need to setup a username and password for this addon to work?

yelhsa083, 2024-05-05:
I found the same, but then saw the descriptor (which now seems to have less detail) User and PW go in the mqtt server line, along with specified port at the end.

mqtt://user:pass@ipaddress:1883
Port assumed to be 1883

SNAPE15, 2024-05-05:
> I found the same, but then saw the descriptor (which now seems to have less detail) User and PW go in the mqtt server line, along with specified port at the end.
> 
> mqtt://user:pass@ipaddress:1883 Port assumed to be 1883

 I knew that mqtt address, but ive never had/created/used a user/pass for mqtt, so asking do I actually need to create that for it to work

yelhsa083, 2024-05-05:
I think you do

Go to mosquito mqtt server
Config
In the options segment there is logins with a typable area

Add to the below to that area and save

`- username: goodusername
  password: goodpassword`

SNAPE15, 2024-05-06:
> I think you do
> 
> Go to mosquito mqtt server Config In the options segment there is logins with a typable area
> 
> Add to the below to that area and save `- username: goodusername` `password: goodpassword`
> 
> Just make sure the alignment has username and password aligned Cannot seem to get it correctly shown here

thanks, got it working. yes, needs that format exact.

now to play....

NickStallman, 2024-05-06:
Let me know how you go. :)

Quite a few addons use MQTT so I didn't provide the exact instructions for setting it up.

MQTT could be anonymous without a login in which case mqtt://hostname should work skipping the user:pass@ bit.
I just realised my example in the README appears to be stuffed so I'll double check why that happened.

NickStallman, 2024-05-08:
If everything is going well with this, I'll close this ticket.
The docs should be slightly clearer now on the mqtt url format.

closed 2024-05-08 by NickStallman

SNAPE15, 2024-05-08:
So far yes, all good. Havent played too much yet.