NickStallman/home-assistant-repo #4: Lower poll_interval?

kalaws, 2024-05-26:
Thanks for the great work!

many chance we could get lower polling interval than 10s? Using the data for load balancing.

NickStallman, 2024-05-28:
I don't actually know what the internal update frequency is of the Winet dongle, so how often the dongle gets new information from the inverter. That'll be the maximum update rate.

I've released a new version which reduces the minimum down to 1 second so you can tinker from here and see what works for you.

I ran mine at an update rate of 2 seconds and it seems it wasn't getting a huge amount of new data. So maybe 3-5 seconds might be the practical limit, but it did work.
```
2024-05-28 20:50:20 info: Updated 10 sensors 
2024-05-28 20:50:24 info: Updated 14 sensors 
2024-05-28 20:50:26 info: Updated 7 sensors 
2024-05-28 20:50:28 info: Updated 2 sensors 
2024-05-28 20:50:30 info: Updated 16 sensors 
2024-05-28 20:50:32 info: Updated 8 sensors 
2024-05-28 20:50:34 info: Updated 1 sensors 
2024-05-28 20:50:36 info: Updated 19 sensors 
2024-05-28 20:50:40 info: Updated 1 sensors 
2024-05-28 20:50:42 info: Updated 21 sensors 
2024-05-28 20:50:46 info: Updated 16 sensors 
```

Just be aware it seems the Winet devices seem to be CPU limited and like to lock up if they get overloaded, so updating faster and doing other things like Modbus communication might not be as stable. Let me know what you find though. :)

kalaws, 2024-05-29:
Thanks! I have the older WiNet-S, it seems. Mine appears to jam quite easily as I go below 5 s. With 3 s, for example, only like 8 sensors update at each poll. I had a Fronius Symo before, where a json file could easily be read every second, with updates for all values on every instance. I find this WiNet-S thing a strange module compared to that...

NickStallman, 2024-06-20:
I'll close this ticket if there's no further issues with poll interval (beyond the module being silly of course). :)

closed 2024-06-20 by NickStallman