NickStallman/home-assistant-repo #5: Test Connect with my SG10RT on Winet-S

hakensline, 2024-05-27:
Hello !

Tanks for your job ! i tested your integration but i have this issue :

####
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started

> cli
> node build/src/index.js

2024-05-27 18:01:31 info: Connected to MQTT broker 
2024-05-27 18:01:32 info: Fetched l18n properties. 
2024-05-27 18:01:32 info: Running a newer firmware version 
2024-05-27 18:01:32 info: Connected to Winet, logging in 
2024-05-27 18:01:32 info: Authenticated successfully 
2024-05-27 18:01:32 error: Invalid devicelist message: {
  data: {
    result_code: 1,
    result_msg: 'success',
    result_data: { service: 'devicelist', list: [Array], count: 1 }
  },
  [Symbol(level)]: 'error',
  [Symbol(splat)]: [ { data: [Object] } ]
}
/usr/src/app/build/src/winetHandler.js:343
            this.currentDevice = this.devices[0].dev_id;
                                                 ^

TypeError: Cannot read properties of undefined (reading 'dev_id')
    at winetHandler.scanDevices (/usr/src/app/build/src/winetHandler.js:343:50)
    at Timeout._onTimeout (/usr/src/app/build/src/winetHandler.js:95:22)
    at listOnTimeout (node:internal/timers:573:17)
    at process.processTimers (node:internal/timers:514:7)

Node.js v20.12.1
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

####

My config: 

winet_host: 192.168.23.238
mqtt_url: mqtt://myuser:mypass@core-mosquitto
winet_user: admin
winet_pass: passforwinets
poll_interval: 10

Tanks for the help !

NickStallman, 2024-05-28:
It looks like you've got the older Winet-S dongle not the newer Winet-S2 dongle. I only physically have a S2 so it's a little tricky to debug but they do seem to be very similar to each other.

Others in #3 are also trying it with the Winet-S and have provided some further debug info, and I've released 0.1.6 which might work wiith the Winet-S. Give that a go and let me know if you still have issues.

NickStallman, 2024-06-20:
The older Winet S seems to be working well now so I'll close this ticket.

closed 2024-06-20 by NickStallman