NickStallman/home-assistant-repo #7: Websocket error

macduff89, 2024-06-08:
So far I've not had the add-on run for more than a couple of days before seeing this error:

`2024-06-05 13:00:28 error: Websocket error: `

It does not trigger the watchdog but stops reading data. As soon as I notice, I restart the add-on and it starts working again, but that might be a day or so later. I assume the dongle is getting overloaded as it has done occasionally when logging in via browser. Is it possible to have it try again after waiting half an hour or so?

Here is a complete log if that helps at all:

`s6-rc: info: service s6rc-oneshot-runner: starting`
`s6-rc: info: service s6rc-oneshot-runner successfully started`
`s6-rc: info: service fix-attrs: starting`
`s6-rc: info: service fix-attrs successfully started`
`s6-rc: info: service legacy-cont-init: starting`
`s6-rc: info: service legacy-cont-init successfully started`
`s6-rc: info: service legacy-services: starting`
`s6-rc: info: service legacy-services successfully started`

> cli
> node build/src/index.js

2024-06-05 11:27:52 info: Connected to MQTT broker 
2024-06-05 11:27:52 info: Fetched l18n properties. 
2024-06-05 11:27:52 info: Connected to a older Winet-S device 
2024-06-05 11:27:52 info: Connected to Winet, logging in 
2024-06-05 11:27:53 info: Authenticated successfully 
2024-06-05 11:27:53 info: Detected device: SG80RS (A2360814641) 
2024-06-05 11:27:53 info: Registered device: SG80RS_A2360814641 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 total_ongrid_running_time 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 daily_yield 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 total_yield 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 running_status 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 bus_voltage 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 internal_air_temperature 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 array_insulation_resistance 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 total_dc_power 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 total_active_power 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 total_reactive_power 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 total_apparent_power 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 total_power_factor 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 grid_frequency 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 phase_a_voltage 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 phase_a_current 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 number_of_afci_faults 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 maximum_apparent_power 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 mppt1_voltage 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 mppt1_current 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 mppt1_power 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 mppt2_voltage 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 mppt2_current 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 mppt2_power 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 mppt3_voltage 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 mppt3_current 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 mppt3_power 
2024-06-05 11:27:53 info: Configured sensor: SG80RS_A2360814641 mppt_total_power 
2024-06-05 11:27:53 info: Configured 27 sensors 
2024-06-05 11:27:53 info: Updated 27 sensors 
2024-06-05 11:33:03 info: Updated 27 sensors 
2024-06-05 11:38:03 info: Updated 27 sensors 
2024-06-05 11:43:10 info: Updated 27 sensors 
2024-06-05 11:48:18 info: Updated 27 sensors 
2024-06-05 11:53:28 info: Updated 27 sensors 
2024-06-05 11:58:34 info: Updated 27 sensors 
2024-06-05 12:03:34 info: Updated 27 sensors 
2024-06-05 12:08:34 info: Updated 27 sensors 
2024-06-05 12:13:34 info: Updated 27 sensors 
2024-06-05 12:18:34 info: Updated 27 sensors 
2024-06-05 12:23:35 info: Updated 27 sensors 
2024-06-05 12:28:37 info: Updated 27 sensors 
2024-06-05 12:33:35 info: Updated 17 sensors 
2024-06-05 12:33:45 info: Updated 10 sensors 
2024-06-05 12:38:36 info: Updated 17 sensors 
2024-06-05 12:38:45 info: Updated 10 sensors 
2024-06-05 12:43:45 info: Updated 27 sensors 
2024-06-05 12:48:45 info: Updated 17 sensors 
2024-06-05 12:48:56 info: Updated 10 sensors 
2024-06-05 12:53:48 info: Updated 17 sensors 
2024-06-05 12:53:58 info: Updated 10 sensors 
2024-06-05 12:59:56 error: Watchdog triggered, reconnecting 
2024-06-05 12:59:56 warn: Reconnecting to Winet 
2024-06-05 13:00:28 error: Websocket error:

macduff89, 2024-06-20:
Installed the latest update a couple of days ago. So far so good!

NickStallman, 2024-06-20:
Great to hear! I'll close this issue but let me know if you have further issues.

closed 2024-06-20 by NickStallman