NickStallman/home-assistant-repo #8: SG20RS - no grid usage or export

mitch-1211, 2024-07-05:
Thanks for this great add on!

I am able to get data coming across into my home assistant instance, however I can't seem to get any info from the power meter i.e. I have no info about grid consumption or daily feed in etc

Below is a list of all the entities that have been created:

```
Sungrow Inverter Meter Phase-A Current
sensor.sg10rs_a23a2000179_meter_phase_a_current
MQTT
—
—

Sungrow Inverter Meter Phase-A Voltage
sensor.sg10rs_a23a2000179_meter_phase_a_voltage
MQTT
—
—

Sungrow Inverter Meter Phase-B Current
sensor.sg10rs_a23a2000179_meter_phase_b_current
MQTT
—
—

Sungrow Inverter Meter Phase-B Voltage
sensor.sg10rs_a23a2000179_meter_phase_b_voltage
MQTT
—
—

Sungrow Inverter Meter Phase-C Current
sensor.sg10rs_a23a2000179_meter_phase_c_current
MQTT
—
—

Sungrow Inverter Meter Phase-C Voltage
sensor.sg10rs_a23a2000179_meter_phase_c_voltage
MQTT
—
—

Sungrow Inverter MPPT Total Power
sensor.sg10rs_a23a2000179_mppt_total_power
MQTT
—
—

Sungrow Inverter MPPT1 Current
sensor.sg10rs_a23a2000179_mppt1_current
MQTT
—
—

Sungrow Inverter MPPT1 Power
sensor.sg10rs_a23a2000179_mppt1_power
MQTT
—
—

Sungrow Inverter MPPT1 Voltage
sensor.sg10rs_a23a2000179_mppt1_voltage
MQTT
—
—

Sungrow Inverter MPPT2 Current
sensor.sg10rs_a23a2000179_mppt2_current
MQTT
—
—

Sungrow Inverter MPPT2 Power
sensor.sg10rs_a23a2000179_mppt2_power
MQTT
—
—

Sungrow Inverter MPPT2 Voltage
sensor.sg10rs_a23a2000179_mppt2_voltage
MQTT
—
—

Sungrow Inverter MPPT3 Current
sensor.sg10rs_a23a2000179_mppt3_current
MQTT
—
—

Sungrow Inverter MPPT3 Power
sensor.sg10rs_a23a2000179_mppt3_power
MQTT
—
—

Sungrow Inverter MPPT3 Voltage
sensor.sg10rs_a23a2000179_mppt3_voltage
MQTT
—
—

Sungrow Inverter Number of AFCI Faults
sensor.sg10rs_a23a2000179_number_of_afci_faults
MQTT
—
—

Sungrow Inverter Phase A Current
sensor.sg10rs_a23a2000179_phase_a_current
MQTT
—
—

Sungrow Inverter Phase A Voltage
sensor.sg10rs_a23a2000179_phase_a_voltage
MQTT
—
—

Sungrow Inverter Running Status
sensor.sg10rs_a23a2000179_running_status
MQTT
—
—

Sungrow Inverter SG10RS A23A2000179
sensor.sg10rs_a23a2000179_sg10rs_a23a2000179
MQTT
—
—

Sungrow Inverter Total Active Power
sensor.sg10rs_a23a2000179_total_active_power
MQTT
—
—

Sungrow Inverter Total Apparent Power
sensor.sg10rs_a23a2000179_total_apparent_power
MQTT
—
—

Sungrow Inverter Total DC Power
sensor.sg10rs_a23a2000179_total_dc_power
MQTT
—
—

Sungrow Inverter Total On-grid Running Time
sensor.sg10rs_a23a2000179_total_on_grid_running_time
MQTT
—
—

Sungrow Inverter Total Power Factor
sensor.sg10rs_a23a2000179_total_power_factor
MQTT
—
—

Sungrow Inverter Total Reactive Power
sensor.sg10rs_a23a2000179_total_reactive_power
MQTT
—
—

Sungrow Inverter Total Yield
sensor.sg10rs_a23a2000179_total_yield
MQTT
—
—
```

So it seems as though the below would give the current going through the meter device, however as you can see it's not changing value at all

[image]

There's no problem getting total solar generation:
[image]

Any idea how to get:
-instantaneous grid usage (or feed in)
-instantaneous solar production
-daily integrated grid usage

Many thanks

NickStallman, 2024-07-05:
The meter readings should be functional.
Note that all the possible meter values will appear, even if they aren't actually used.
This applies to other sensors too, not all are expected to actually be usable - they are just what's provided as-is.

Here's what I get, only Phase A is used with my setup so B and C are ignored.

[image]

Do you see the meter readings if you go to the Winet's web interface directly?
Do you see meter readings in iSolarCloud? It sounds like your meter might not be hooked up properly if they don't show the values.

There's notes in the Readme for which values give the various instantaneous readings, and how to get the Energy Dashboard working.
All the values should be available, even if the names don't seem obvious initially. :)
I'm not sure if the SG20RS changes things or not (since the addon just reads all the data available from the Winet) but I haven't had any reports that these instructions differ for other inverters.

mitch-1211, 2024-07-05:
Firstly, apologies, I mistakenly wrote the wrong model in the title originally. Correct model is SG10RS

So looking in the iSolar Cloud app, I get information about both grid usage and solar production. 

I’ll put a few screenshots of the individual devices and the data they are producing:

[image]

[image]
[image]
[image]
[image]

So it looks like the meter device itself is producing a single value that is positive for grid usage and negative for grid feed in. 

The inverter device doesn’t seem to have the meter data associated with it directly. It seems as though this info is missing:
[image]

Obviously iSolar cloud is able to piece together the right info to calculate all metrics, just don’t know if the data from the meter is coming through to home assistant. 

Would you mind sharing what your devices look like in isolar cloud in terms of data?

NickStallman, 2024-07-06:
I know there's a few users of my addon with a SG10RS however I don't know of any specific quirks with that setup.

Can you log in to the Winet via it's IP address in your browser and just confirm that you don't see any "Purchased Power" orr "Total Export Active Power" value?
This addon should simply mirror exactly what the Winet web interface shows, but worth double checking.
They are the fields provided by the SH10RS that give the realtime import/export info. I don't actually use the meter values myself.

If iSolarCloud shows "Meter A Active Power" as blank then that sounds like the meter isn't wired to the inverter correctly.
That would explain why Home Assistant also is getting blank. It does look like it knows there's a meter there though, just no values for it.

If the meter isn't wired up I'm not sure how the Inverter would be getting the household power usage - the inverter on it's own can only see generation but not whether it's exported or self consumed.

One thing to double check is are you on the latest firmware?
They do fiddle with these things and change them around.

matheuscanela, 2024-07-06:
Hey @NickStallman 
Thanks for the integration, I just installed and it worked perfectly. 

Do you know why ISolar shows one shows a different load compared to what the inverter produces? I checked the output from the WiNet internal site and it matches what the integration collect, however, the ISolar always shows a little bit above - see the screenshots

[image]
[image]

matheuscanela, 2024-07-06:
This is the graph for the morning, and there is no 1KW consumption.

[image]

NickStallman, 2024-07-07:
Do you have whole house backup or are there loads not on your backup circuit? That sensor isn't your household usage it's backup only.

I calculate the household usage (and I think iSolarCloud does too) from:
- purchased_power (Grid import)
- mppt_total_power (Solar production)
- battery_discharging_power
- battery_charging_power
- total_export_active_power (Grid export)
If you add/subtract them all together correctly, you get your total household usage. It does occasionally have little blips when some sensor values update at different times to others which is annoying but that's to do with how the Winet dongle gets updates.

matheuscanela, 2024-07-07:
Interesting!
It is supposed to be the entire house, so don't know why it is less. I will switch off the circuit outside and see if everything is connected. This inverter supports up to 63A on the backup line, so I connected everything as I can decide what to use during a blackout.

Also, thanks for the calculation tip, it makes totally sense.

mitch-1211, 2024-07-15:
> I know there's a few users of my addon with a SG10RS however I don't know of any specific quirks with that setup.
>
> Can you log in to the Winet via it's IP address in your browser and just confirm [...]

Hi @NickStallman, so can confirm that your add-on is mirroring what is shown on the Winet-S dashboard. Unfortunately for me this means I can't get any instantaneous generation/usage information.

I have setup [SunGather](https://github.com/bohdan-s/SunGather) and it seems to working well so far.

Cheers

NickStallman, 2024-07-18:
@mitch-1211 your Winet-S dashboard isn't showing any instantaneous numbers? That would be very strange.
Check the numbers carefully as I've noticed some of the names are odd/inconsistent, so you might find the correct value under a different name.
E.g. I get imported power as "purchased_power" and exported power as "total_export_active_power". This took me a bit to figure out.

If there's absolutely no figures for instantaneous power that you can find, double check you are on the latest Winet firmware, and if still nothing then unfortunately yeah this addon might not be suitable for you.

IzSmithh, 2024-09-14:
@NickStallman Hello i'm new hear just signed up to see if there have been any updates with this issue?
I'm in the situation as mitch-1211 but unfortunately not as tech savvy to install other complex add-ons. So, this was perfect for me. Thank you for that.
I've got the same Sg10rs inverter. winet s dongle. the extractor reads 34 entities none of them deflect the direction of flow to the grid. I also see that my meter phase A is sitting at 0.01A constantly. and nothing obvious on the winet s webpage too.
In either case look like ill still hold on to this hoping it would get fixed.
Thanks once more.

NickStallman, 2024-09-16:
Check out total_load_active_power. I only noticed recently that it appears to be the household number.
The names aren't very clear and I think in the past I was mixing it up with total active power which iis different.

NickStallman, 2024-09-16:
@IzSmithh Do you have purchased_power and total_export_active_power? That's your grid import and export which gives you direction. Looking at the meter values I've found to be not all that useful.

IzSmithh, 2024-09-17:
Hi Nick, no it doesn't. one useful thing i got was solar produced (total active power) 
[image]
According to sungrow support "SG10RS does not support forwarding the current reading from the meter for either Modbus or Web-API. However, meter's active power data is readable, but only from the Modbus 5083 register."
This led me to install mkaiser implementation with modified or shortened files from https://gist.github.com/nickw444/9a048bc5b6d36f00c869458121c19e86 and it seem to work. (Just done it).
Thank you once more for your addon and great work.
It's really appreciated.

NickStallman, 2024-09-18:
If you are looking for any value in that screenshot, you should definitely find it with this addon - it pulls the exact same data that you see there.
Specifically it pulls everything from the "Realtime Values" and "DC Info" tabs

But yeah a mix of options seems best at the moment, e.g. Mkaiser gives a couple of extra things like export limits and modes, and allows changing values/modes as well which can't easily be done via the web interface API.

av3nger, 2024-09-19:
SG8.0RS inverter. Similar to @IzSmithh and @mitch-1211 - no data on current grid consumption/export

IzSmithh, 2024-09-19:
> SG8.0RS inverter. Similar to @IzSmithh and @mitch-1211 - no data on current grid consumption/export

Sungrow expressed “ The grid feed-in or purchase statistics (daily or total) are uploaded by the meter while daily data is based on the deduction of the previous day's total data from today's total data, but these Modbus registers are not available yet to third-party devices/platforms. They will be available soon via the future firmware updates.”

I don’t know how to code but was going to spend time this weekend trying to figure it out.  
if anyone can help would be much appreciated.

iankds05, 2024-09-23:
I have been using the SunGather repo in Home Assistant to get basic info from my recently installed SG10RS inverter but he gathered entities are pretty limited so I came across your repo and thought I would give it a try. Its installed correct and configured for my mqtt broker but there are no new entities showing either in the devices tag  or in mqtt . I am currently running HA 2024.9.2 in a VM and todate had no config issues with either the SunGather or Gosungrow repos. The logs seem to show everything has started ok but having installed it several times it still does not show any gathered entities

2024-09-23 03:29:58 warn: Reconnecting to Winet 
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started

> cli
> node build/src/index.js

2024-09-23 03:30:14 info: Connected to MQTT broker 
2024-09-23 03:30:15 info: Fetched i18n properties. 
Could you suggest anything to look for . If you need any more info just let me know

NickStallman, 2024-09-29:
@iankds05 what Winet IP address are you using? When you go to https://192.168.1.x (where that is your Winet's IP address) what do you get (past the SSL warning).
I'd expect something after "Fetched i18n properties." whether it's a error or success.

iankds05, 2024-09-29:
Tanks for your reply. Firstly the initial issue was that due to the pretty hopeless WIFI reception range of the WiNet dongle I was getting a lot of drop outs. My inverter is installed on the outside North facing wall with my internet router at the far end of the South wall about 50m distance through 2 brick walls. Even with an indoor signal repeater the reception is not good so I have bitten the bullet and installed a Lan cable to the router. I now get some readings but similar to @IzSmithh I don't have purchased or exported power readings, and am having to use a bi-directional CT clamp to get these values. The S100 meter values displayed are only useful as a reference. Attached is a screen shot of mqtt sensors found listing

[image]

NickStallman, 2024-09-29:
Ah yep spotty wifi would do it. This addon keeps a persistent WebSocket connection open to stream the data.

Must be something to do with the different inverters. Sungrow is notoriously inconsistent between different models.
Unfortunately if the numbers aren't directly visible in the browser when accessing the Winet, my addon cannot show it.

I'll close this issue as it sounds like the SG10RS just won't supply the numbers via web interface.

closed 2024-09-29 by NickStallman

iankds05, 2024-09-29:
Ok Nick thanks anyway. I guess its not easy to figure out Chinese equipment with no standards adopted by them for data transmission. Appreciate you time on this matter and fine to close it out