NickStallman/home-assistant-repo #9: multi device support

SwissFlitzer, 2024-07-10:
Hi
Is it possible to add multiple devices to WiNet extractor / HA Addon?

dbuggz, 2024-07-25:
I have two inverters, an SG5 and SH5. can i add the secons via the GUI or editing .yaml?
Dave

NickStallman, 2024-09-18:
How are you guys running this addon? Inside of Home Assistant or as a standalone docker image?
And do your multiple Winet's use the same username and password? Or must they have different passwords?

SwissFlitzer, 2024-09-19:
> How are you guys running this addon? Inside of Home Assistant or as a standalone docker image? And do your multiple Winet's use the same username and password? Or must they have different passwords?

Thanks for reply. 

Addon is running inside of HA. Username & PW is the same for every device.

mcvax, 2024-09-19:
Thanks Nick
I'm running it inside HA and using the same username and password on both devices.
Cheers
Chris

Tobias-N7, 2024-11-16:
The same for me, would be nice to get data from the slave inverter, without running something outside from HA.

Thanks

NickStallman, 2026-09-26:
Multi WiNet support is finally here in 0.3.0 :)

You can now list several WiNets in the one addon by separating them with commas in **Winet Host**, e.g. `192.168.1.100, 192.168.1.101`. They all use the same username/password, which from this thread sounds like what everyone has. If one needs different credentials there's also an optional **Additional WiNets** list where you can override the user/pass/ssl per dongle.

Existing entities aren't renamed, each inverter just shows up as its own device. With two or more inverters you also get a "WiNet Site" device that adds up PV power and daily/lifetime PV yield across them all, which is handy for the Energy dashboard.

I'm running it here with an SH10RS on one WiNet-S2 and an SG5.0RS on another. More details in the README. Let me know how you go.

closed 2026-09-26 by NickStallman