NickStallman/home-assistant-repo #11: SH15T ends up with a semi-translated entity names

ctheune, 2024-07-19:
On startup it shows this:

```
2024-07-19 08:47:36 info: Configured sensor: SH15T_ i18n_common_group_bunch_title_andpercent1_voltage 
2024-07-19 08:47:36 info: Configured sensor: SH15T_ i18n_common_group_bunch_title_andpercent1_current 
2024-07-19 08:47:36 info: Configured sensor: SH15T_ i18n_common_group_bunch_title_andpercent1_power 
2024-07-19 08:47:36 info: Configured sensor: SH15T_ i18n_common_group_bunch_title_andpercent2_voltage 
2024-07-19 08:47:36 info: Configured sensor: SH15T_ i18n_common_group_bunch_title_andpercent2_current 
2024-07-19 08:47:36 info: Configured sensor: SH15T_ i18n_common_group_bunch_title_andpercent2_power 
2024-07-19 08:47:36 info: Configured sensor: SH15T_ i18n_common_group_bunch_title_andpercent3_voltage 
2024-07-19 08:47:36 info: Configured sensor: SH15T_ i18n_common_group_bunch_title_andpercent3_current 
2024-07-19 08:47:36 info: Configured sensor: SH15T_ i18n_common_group_bunch_title_andpercent3_power 
2024-07-19 08:47:36 info: Configured sensor: SH15T_ i18n_common_group_bunch_title_andpercent4_voltage 
2024-07-19 08:47:36 info: Configured sensor: SH15T_ i18n_common_group_bunch_title_andpercent4_current 
2024-07-19 08:47:36 info: Configured sensor: SH15T_ i18n_common_group_bunch_title_andpercent4_power 
2024-07-19 08:47:36 info: Configured sensor: SH15T_ i18n_common_group_bunch_title_andpercent5_voltage 
2024-07-19 08:47:36 info: Configured sensor: SH15T_ i18n_common_group_bunch_title_andpercent5_current 
2024-07-19 08:47:36 info: Configured sensor: SH15T_ i18n_common_group_bunch_title_andpercent5_power 
2024-07-19 08:47:36 info: Configured sensor: SH15T_ i18n_common_group_bunch_title_andpercent6_voltage 
2024-07-19 08:47:36 info: Configured sensor: SH15T_ i18n_common_group_bunch_title_andpercent6_current 
2024-07-19 08:47:36 info: Configured sensor: SH15T_ i18n_common_group_bunch_title_andpercent6_power 
```

Other entities are fine, though.

[image]

NickStallman, 2024-07-19:
That's very interesting. Is your Winet configured for English or a different language by any chance?

When you go to the Winet interface directly, can you find those values and see what name it gives them?
"group_bunch_title_andpercent3" is rather odd and doesn't describe much.

wangeris, 2024-08-27:
I'm getting something similar on SG10RT 

[image]

ctheune, 2024-09-06:
I sniffed the network connection and found the cause of this:

Here's the result that includes the relevant strings:

```
	"result_data":	{
		"service":	"direct",
		"list":	[{
				"name":	"MPPT1",
				"voltage":	"734.6",
				"voltage_unit":	"V",
				"current":	"6.0",
				"current_unit":	"A"
			}, {
				"name":	"MPPT2",
				"voltage":	"759.7",
				"voltage_unit":	"V",
				"current":	"3.0",
				"current_unit":	"A"
			}, {
				"name":	"MPPT3",
				"voltage":	"0.0",
				"voltage_unit":	"V",
				"current":	"0.0",
				"current_unit":	"A"
			}, {
				"name":	"I18N_COMMON_GROUP_BUNCH_TITLE_AND%@1",
				"voltage":	"734.6",
				"voltage_unit":	"V",
				"current":	"6.00",
				"current_unit":	"A"
			}, {
				"name":	"I18N_COMMON_GROUP_BUNCH_TITLE_AND%@2",
				"voltage":	"734.6",
				"voltage_unit":	"V",
				"current":	"0.00",
				"current_unit":	"A"
			}, {
				"name":	"I18N_COMMON_GROUP_BUNCH_TITLE_AND%@3",
				"voltage":	"759.7",
				"voltage_unit":	"V",
				"current":	"3.00",
				"current_unit":	"A"
			}, {
				"name":	"I18N_COMMON_GROUP_BUNCH_TITLE_AND%@4",
				"voltage":	"759.7",
				"voltage_unit":	"V",
				"current":	"0.00",
				"current_unit":	"A"
			}, {
				"name":	"I18N_COMMON_GROUP_BUNCH_TITLE_AND%@5",
				"voltage":	"0.0",
				"voltage_unit":	"V",
				"current":	"0.00",
				"current_unit":	"A"
			}, {
				"name":	"I18N_COMMON_GROUP_BUNCH_TITLE_AND%@6",
				"voltage":	"--",
				"voltage_unit":	"V",
				"current":	"--",
				"current_unit":	"A"
			}],
		"count":	9
	}
}
```

ctheune, 2024-09-06:
The UI when logging into Winet is English. The system is running in Germany and configured for the German grid. But I think this isn't relevant as the weird strings appear directly in the output ...

NickStallman, 2024-09-12:
It will likely be related to using German in the Web UI actually, as the addon currently loads the internal Winet US language file to get the correct field names.

I have no clue why the German interface would be showing values which aren't shown in the English interface though. That's very weird.

Can you find in the Web UI where "I18N_COMMON_GROUP_BUNCH_TITLE_AND" might be coming from and take a screenshot?
Then maybe change the UI to English and see what the same area shows?

Tobias-N7, 2024-11-16:
Hi, 

I have exactly the same problem. Those look like the DC string values which aren't translated between the English and German version of the web interface.

[image]
[image]

NickStallman, 2024-11-29:
Thanks @Leahcim70 

This tweak is released in version 0.2.2 so give it a go and hopefully this makes it work better for the SH15T.

Tobias-N7, 2024-11-29:
This looks really good! The values are now showing up with the right labels.

Thank you for the fast fix! 

I hope that my Panels are get installed in the next few weeks. Until then I only see zeros.

`2024-11-29 21:35:07 info: Configured sensor: SH15T_A2430200418 string_1_voltage 
2024-11-29 21:35:07 info: Configured sensor: SH15T_A2430200418 string_1_current 
2024-11-29 21:35:07 info: Configured sensor: SH15T_A2430200418 string_1_power 
2024-11-29 21:35:07 info: Configured sensor: SH15T_A2430200418 string_2_voltage 
2024-11-29 21:35:07 info: Configured sensor: SH15T_A2430200418 string_2_current 
2024-11-29 21:35:07 info: Configured sensor: SH15T_A2430200418 string_2_power 
2024-11-29 21:35:07 info: Configured sensor: SH15T_A2430200418 string_3_voltage 
2024-11-29 21:35:07 info: Configured sensor: SH15T_A2430200418 string_3_current 
2024-11-29 21:35:07 info: Configured sensor: SH15T_A2430200418 string_3_power 
2024-11-29 21:35:07 info: Configured sensor: SH15T_A2430200418 string_4_voltage 
2024-11-29 21:35:07 info: Configured sensor: SH15T_A2430200418 string_4_current 
2024-11-29 21:35:07 info: Configured sensor: SH15T_A2430200418 string_4_power 
2024-11-29 21:35:07 info: Configured sensor: SH15T_A2430200418 string_5_voltage 
2024-11-29 21:35:07 info: Configured sensor: SH15T_A2430200418 string_5_current 
2024-11-29 21:35:07 info: Configured sensor: SH15T_A2430200418 string_5_power 
2024-11-29 21:35:07 info: Configured sensor: SH15T_A2430200418 string_6_voltage 
2024-11-29 21:35:07 info: Configured sensor: SH15T_A2430200418 string_6_current 
2024-11-29 21:35:07 info: Configured sensor: SH15T_A2430200418 string_6_power `

[image]

Leahcim70, 2024-11-30:
> Thanks @Leahcim70
> 
> This tweak is released in version 0.2.2 so give it a go and hopefully this makes it work better for the SH15T.

Hi Nick, 

i hope that i did not break anything on other devices. Are there vales to be added that are not named mppt* ?

Grüße
Michael

NickStallman, 2024-11-30:
Looks good on my SH10RS. I don't think other inverters use the numeric placeholders like that. :)

closed 2025-01-09 by NickStallman