NickStallman/home-assistant-repo #14: WiNet-S2 automatic logout prompt ends login session in proportion to polling frequency

AuThomasPrime, 2024-08-23:
When the WiNet extractor addon is running, the WiNet-S2 web portal will constantly present you with an automatic logout prompt as frequently as the set polling interval. This makes the web portal unusable at low polling intervals (e.g. <= 10 seconds). I need to turn off the addon if I want to use the web portal effectively.

[image]

wangeris, 2024-08-23:
I'm getting the same, but I believe it's a limitation of the WiNet adapter permitting singular connection (due to limited resources) rather than a problem with WiNet extractor itself.

AuThomasPrime, 2024-08-23:
That makes sense. If it's not causing any other issues then I think it's a small price to pay for having access to the data in Home Assistant. Once setup, the web portal isn't something you really need to access frequently anyway.

NickStallman, 2024-09-12:
Yep this effect is expected with the newest firmware. The Winet only wants one user accessing it at a time essentially.
Earlier firmware may have allowed it, but it did cause some rather unpredictable behaviour which is probably why they put the restriction in.

I find myself very rarely needing the UI, and when I do I just stop the addon then start it again once I'm done.

I don't believe there's any possible bypass, and certainly not a stable one.

closed 2024-09-12 by NickStallman