NickStallman/home-assistant-repo #15: MQTT Timeout

moku01, 2024-08-23:
Hey, can somebody tell me whats wrong with my confic?

[image]

 I'm always getting that erro message... :(

NickStallman, 2024-09-09:
I've never seen that with my set up.

Can you tell me how you've configured MQTT? Where is it running?

NickStallman, 2024-09-09:
So just Mosquito MQTT as an addon? And MQTT Explorer on your PC connects fine?

What host/IP do you have configured in the MQTT connection string? Have you configured the Pi's IP just like how you connect with MQTT Explorer?

moku01, 2024-09-13:
Yes MQTT runs on the Pi, i can access ist via the HomeAssistant MQTT >Explorer Ad-On as well as the Desktop version. The Host IP is, as far as i know the Pi's IP Adress on port 1883

NickStallman, 2024-09-18:
I've done some googling and there's a few results with connack timeouts but nothing specific.

Is your MQTT server particularly busy and/or your Pi's CPU usage particularly high?
That's the only thing I can think of. Most search results seem to indicate high traffic on the MQTT side being the cause, rather than something specific this addon is doing.

moku01, 2024-09-20:
Thank you for your help, but the CPU is just used aroud 10% and no other Device uses MQTT. 
I'm gonna try to find a different solution.

NickStallman, 2024-09-29:
No worries. I'll close this issue as I'm not sure it's an issue with this addon specifically.

closed 2024-09-29 by NickStallman