NickStallman/home-assistant-repo #17: Mqtt wrong - tried several formats but nothing works

cryd-s, 2024-08-30:
[image]

I tried:

mqtt://mqttuser:"topsecret#001"@192.168.1.10:1883
mqttuser:"topsecret#001"@192.168.1.10:1883
mqtt://mqttuser:"topsecret#001"@192.168.1.10
mqttuser:"topsecret#001"@192.168.1.10

cryd-s, 2024-08-30:
the credentials are correct

42asdf, 2024-09-04:
I think you cannot give username and password in the mqtt URL. I implemented options to give the MQTT username and password for #18  and opened a pull request.

NickStallman, 2024-09-09:
@cryd-s I wonder if you have any special characters in your password which may be throwing it off?

getaddrinfo is trying to resolve mqttuser which is of course not right.
The format ```mqtt://<username>:<password>@<host>``` works well for me, however my password is very long and  doesn't include any symbols.

42asdf, 2024-09-10:
If you have special characters in your password, you can try encoding them e.g. using the following python code:
```
import urllib.parse
password = "a%€$"
print(urllib.parse.quote(password, safe=''))
```

NickStallman, 2024-09-18:
Yep a urlencode will sort out any complex passwords, an easy approach is a online tool like https://www.urlencoder.org/

closed 2024-09-18 by NickStallman