NickStallman/home-assistant-repo #18: Add option to give username and password for mqtt

42asdf, 2024-09-01:
Currently only the MQTT_URL can be provided. However, if the mqtt broker is password protected, a username and password needs to be given. Otherwise errors like `MQTT error: ErrorWithReasonCode: Connection refused: Not authorized` appear.

NickStallman, 2024-09-09:
MQTT does already support a username and password, it's given directly in URL format as per ```mqtt://<username>:<password>@<host>```

42asdf, 2024-09-10:
I still think it's better to give the username and password not as part of the url, as you don't need to escape special characters in this case. Additionally, you can use a different handling for the password (e.g. access rights, not commit to git, password field in UI).

NickStallman, 2024-09-18:
It would be a minor improvement, but considering it works fine as-is and there's already a lot of users using the url format and it would be a pain to make this backwards compatible, I don't really see this improvement as worth the trouble.

closed 2024-09-18 by NickStallman