NickStallman/home-assistant-repo #21: Sungrow Icloud login issues

brodiep, 2024-09-09:
I am having issues with the add-on logging on to Icloud, I can login to Icloud on my phone app and my desktop without issue and when I user these credentials in the config for the add-on I seem to be having problems, could you please review logs and comment, it may be something I am missing.
### Add-on **log**
Node.js v20.15.1
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started

> cli
> node build/src/index.js

2024-09-09 00:46:53 info: Connected to MQTT broker 
2024-09-09 00:46:53 info: Fetched l18n properties. 
2024-09-09 00:46:53 info: Connected to a Winet-S2 device with older firmware 
2024-09-09 00:46:53 info: Connected to Winet, logging in 
2024-09-09 00:46:53 error: Invalid login message: {
  data: {
    result_code: 102,
    result_msg: 'I18N_COMMON_ACCOUNT_PSSWD_TIP',
    result_data: { service: 'login' }
  },
  [Symbol(level)]: 'error',
  [Symbol(splat)]: [ { data: [Object] } ]
}
/usr/src/app/build/src/winetHandler.js:374
            this.currentDevice = this.devices[0].dev_id;
                                                 ^

TypeError: Cannot read properties of undefined (reading 'dev_id')
    at winetHandler.scanDevices (/usr/src/app/build/src/winetHandler.js:374:50)
    at Timeout._onTimeout (/usr/src/app/build/src/winetHandler.js:97:22)
    at listOnTimeout (node:internal/timers:573:17)
    at process.processTimers (node:internal/timers:514:7)

Node.js v20.15.1
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

### **_MQTT log_**
Node.js v20.15.1
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started

> cli
> node build/src/index.js

2024-09-09 00:46:53 info: Connected to MQTT broker 
2024-09-09 00:46:53 info: Fetched l18n properties. 
2024-09-09 00:46:53 info: Connected to a Winet-S2 device with older firmware 
2024-09-09 00:46:53 info: Connected to Winet, logging in 
2024-09-09 00:46:53 error: Invalid login message: {
  data: {
    result_code: 102,
    result_msg: 'I18N_COMMON_ACCOUNT_PSSWD_TIP',
    result_data: { service: 'login' }
  },
  [Symbol(level)]: 'error',
  [Symbol(splat)]: [ { data: [Object] } ]
}
/usr/src/app/build/src/winetHandler.js:374
            this.currentDevice = this.devices[0].dev_id;
                                                 ^

TypeError: Cannot read properties of undefined (reading 'dev_id')
    at winetHandler.scanDevices (/usr/src/app/build/src/winetHandler.js:374:50)
    at Timeout._onTimeout (/usr/src/app/build/src/winetHandler.js:97:22)
    at listOnTimeout (node:internal/timers:573:17)
    at process.processTimers (node:internal/timers:514:7)

Node.js v20.15.1
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Any help would be appreciated

Paul

NickStallman, 2024-09-09:
Hi @brodiep this addon is not talking to iSolarCloud at all.

Rather it is 100% local, and needs to talk to your Winet dongle directly.
The username and password referenced is the details set when you go to the Winet's IP address directly in your browser.

brodiep, 2024-09-09:
Thank you for the quick reply, I just assumed that it was via Icloud but now I think about it it should have been obvious. It is all working since I changed the login ID and password, thanks for pointing me in the right direction and thank you for your work, works well.
Paul

NickStallman, 2024-09-10:
Excellent, great to hear. :)

closed 2024-09-10 by NickStallman