NickStallman/home-assistant-repo #24: Control discharge rate and limits (feature request)

Staacks, 2024-10-01:
_First of all, thank you very much for this add-on. It was so refreshing to just install it and to have the inverter integrated into HA within minutes after the installers left without having to bother with manufacturer's restrictions and limiterd cloud access. (Also, hooray for Sungrow on that matter.)_

Now, I was wondering if there is any chance of adding control functions? I am thinking of controlling the discharge rate as well as the charge/discharge limits of the battery (which I do not yet have but am about to add to my system in a few months).

If you are wondering about the use-case:
I have a setup with two different electricity tariffs. The house uses a regular tariff, but my heatpump uses a discounted one. My inverter can power both systems, but will prioritize the regular tariff first and it will only power the heatpump if there is excess power. If I add a battery and if it would not last through the night after a cloudy day, I might end up with the heatpump draining the battery and saving on the cheap tariff while the rest of the house then consumes power on the expensive tariff through the rest of the night. So, limiting the discharge rate such that the battery would spread its charge until sunrise would be an easy way to save a few bucks.

NickStallman, 2024-10-01:
I've thought about adding control functionality, but currently it's out of scope for this addon.

This addon uses the internal Websocket protocol of the Winet which is excellent for extracting most data.
It doesn't seem to have any control functions however so it can't be used to change settings.

Changing settings needs to be done via Modbus, and the excellent Mkaiser repo handles this very nicely with pre-made dashboards: https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant
Personally I just disable the Modbus registers for the data that this addon can get, and just use it for controlling settings.
Using Modbus to get all the data seems to be slightly buggier than websockets, and doesn't have all the data for some inverters like the SH10RS, so mixing the two seems to work really well.

closed 2024-10-01 by NickStallman

Staacks, 2024-10-01:
Thanks for the feedback. I stayed away from modbus because of MPPT3 being unavailable and did not think about mixing both methods. Thanks for the hint, I will try that.