NickStallman/home-assistant-repo #26: Websocket error: read ECONNRESET

iankds05, 2024-10-11:
Hi I have a SG10RS installed and at the moment use the Gosungrow integration. I was wanting to reduce reliance on the Icloud data and just keep data collection local so chose your repo. Installed OK with connection to MQTT broker ok but following errors show in the log. In the config I have just set the inverter IP address and MQTT url for my system. Can you suggest possible fix at all or maybe something to go and tweak. As far as I am aware the WINET-S has not had the default User/Password changed so its left just blank

LOG:
2024-10-11 21:33:08 info: Connected to MQTT broker 
2024-10-11 21:33:09 info: Fetched i18n properties. 
2024-10-11 21:33:09 error: Websocket error: read ECONNRESET {
  errno: -104,
  code: 'ECONNRESET',
  syscall: 'read',
  stack: 'Error: read ECONNRESET\n' +
    '    at TCP.onStreamRead (node:internal/stream_base_commons:218:20)',
  [Symbol(level)]: 'error',
  [Symbol(splat)]: [
    Error: read ECONNRESET
        at TCP.onStreamRead (node:internal/stream_base_commons:218:20) {
      errno: -104,
      code: 'ECONNRESET',
      syscall: 'read'
    }
  ]
}
2024-10-11 21:33:09 warn: Reconnecting to Winet

NickStallman, 2024-10-12:
Weird is this occurring consistently? Connection Reset means the connection got killed at the networking layer.
If it happened as a one off I'd ignore it, but if it happens consistently then that's a bit weird.

Do you have a Winet-S or a Winet-S2? And are you on the latest firmware for it (which should be fully supported)?

DanielTemby, 2024-10-14:
Hi Nick, I am getting a similar issue, have a WiNet-S2 and latest firmware as system was only installed last month

Any ideas on how to resolve

2024-10-13 22:46:20 error: Is the Winet IP address correct? 
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started

> cli
> node build/src/index.js

2024-10-13 22:52:53 info: Connected to MQTT broker 
2024-10-13 22:52:54 error: Failed to fetch l18n properties required to start. aborted {
  code: 'ECONNRESET',
  stack: 'Error: aborted\n' +
    '    at TLSSocket.socketCloseListener (node:_http_client:460:19)\n' +
    '    at TLSSocket.emit (node:events:531:35)\n' +
    '    at node:net:338:12\n' +
    '    at TCP.done (node:_tls_wrap:657:7)',
  [Symbol(level)]: 'error',
  [Symbol(splat)]: [
    Error: aborted
        at TLSSocket.socketCloseListener (node:_http_client:460:19)
        at TLSSocket.emit (node:events:531:35)
        at node:net:338:12
        at TCP.done (node:_tls_wrap:657:7) {
      code: 'ECONNRESET'
    }
  ]
}
2024-10-13 22:52:54 error: Is the Winet IP address correct?

NickStallman, 2024-10-14:
Are you on the latest version of the addon? It should be 0.2.0.

Can you access the Winet in a web browser via http://192.168.1.x or does it only work with https://192.168.1.x?
A recent firmware update made https mandatory and a few other minor tweaks which 0.2.0 handles correctly.

DanielTemby, 2024-10-14:
Yes I am on 0.2.0

When I access with a web browser via HTTPS the response I get is “network connection was lost”

NickStallman, 2024-10-14:
Well that's not good. That would be the same problem, so it's not the addon. :)
With the latest firmware, http should be flaky but https should work perfectly with a security warning.

Perhaps try rebooting the Winet module? You can do this via the iSolarCloud app under Settings -> Configuration Parameters -> Remote Device Restart.

DanielTemby, 2024-10-15:
All working now just needed a firmware upgrade and a restart. Now I just need to configure Energy Dashboards

DanielTemby, 2024-10-16:
Hi Nick,  all working except Daily Purchased Energy and Daily Feed In Energy are “unknown”. Is there another sensor that has these values ? My inverter is a SH10RS

baylockzathras, 2024-10-18:
> Hi Nick, all working except Daily Purchased Energy and Daily Feed In Energy are “unknown”. Is there another sensor that has these values ? My inverter is a SH10RS

I seem to have the same issue, however you can use the total option instead, and HA is able to work some Voodoo to get what it needs from that for the energy board

NickStallman, 2024-11-02:
Great, yeah there's some things that the Winet has but without any values, and Home Assistant helpers can work around any gaps.

closed 2024-11-02 by NickStallman