NickStallman/home-assistant-repo #31: Installed but not sure of next steps

MRIC8970, 2024-11-13:
I seem to have installed successfully, maybe. Log shows:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started

> cli
> node build/src/index.js

2024-11-13 08:49:36 info: Fetched i18n properties. 
2024-11-13 08:49:37 info: Connected to a Winet-S2 device with newer firmware 
2024-11-13 08:49:37 info: Connected to Winet, logging in 
2024-11-13 08:49:37 info: Authenticated successfully 
2024-11-13 08:49:38 info: Detected device: SG80RS (A2480211632) 
2024-11-13 08:49:38 info: Updated 32 sensors 
2024-11-13 08:49:38 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:49:39 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:49:40 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:49:41 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:49:42 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:49:44 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:49:45 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:49:46 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:49:47 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:49:48 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:49:49 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:49:50 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:49:51 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:49:52 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:49:53 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:49:54 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:49:55 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:49:56 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:49:57 info: Updated 1 sensors 
2024-11-13 08:49:57 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:49:58 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:49:59 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:50:00 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:50:01 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 
2024-11-13 08:50:02 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 

but I don't understand the next step.

When I try to add stats in Energy -> Add Grid Consumption for example, I'm told I don't have any statistics.

I'm guessing I'm missing some final steps, but can't see in the instructions what I need to do next.

Really keen to get this working--great idea and will really fill a huge gap with these inverters.  Thank you!

Sorry but new to all this, so likely just noob issues

NickStallman, 2024-11-13:
You need to check your MQTT connection details, you should configure it with a url like "mqtt://username:password@ip".

You've connected to the Winet module correctly, but not to MQTT which is how the metrics get sent to Home Assistant.

MRIC8970, 2024-11-14:
Thank you!  Resolved. It wasn't obvious to me that '...@ip' was the Home Assistant IP and MQTT port, but got there in the end. 

Any idea why I don't seem to have any sensor that equates to 'Grid Consumption: daily_purchased_energy'?  This is what I have available:

2024-11-14 05:10:40 info: Connected to MQTT broker 
2024-11-14 05:10:41 info: Fetched i18n properties. 
2024-11-14 05:10:42 info: Connected to a Winet-S2 device with newer firmware 
2024-11-14 05:10:42 info: Connected to Winet, logging in 
2024-11-14 05:10:43 info: Authenticated successfully 
2024-11-14 05:10:43 info: Detected device: SG80RS (A2********2) 
2024-11-14 05:10:43 info: Registered device: SG80RS_A2********2 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 total_ongrid_running_time 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 daily_yield 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 total_yield 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 running_status 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 bus_voltage 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 internal_air_temperature 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 array_insulation_resistance 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 total_dc_power 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 total_active_power 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 total_reactive_power 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 total_apparent_power 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 total_power_factor 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 grid_frequency 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 phase_a_voltage 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 phase_a_current 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 maximum_apparent_power 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 meter_phasea_voltage 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 meter_phaseb_voltage 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 meter_phasec_voltage 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 meter_phasea_current 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 meter_phaseb_current 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 meter_phasec_current 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 mppt1_voltage 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 mppt1_current 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 mppt1_power 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 mppt2_voltage 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 mppt2_current 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 mppt2_power 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 mppt3_voltage 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 mppt3_current 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 mppt3_power 
2024-11-14 05:10:43 info: Configured sensor: SG80RS_A2********2 mppt_total_power 
2024-11-14 05:10:43 info: Configured 32 sensors 
2024-11-14 05:10:43 info: Updated 32 sensors 

Thanks again!

NickStallman, 2024-11-14:
Good to hear you got it working. :)

closed 2024-11-14 by NickStallman