NickStallman/home-assistant-repo #32: power_factor -- no unit of measurement

mtrekker, 2024-11-13:
Is this config or winet-extractor issue ?

```
2024-11-13 12:13:03.951 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.sh15t_**********_total_power_factor (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement '' which is not a valid unit for the device class ('power_factor') it is using; expected one of ['no unit of measurement', '%']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
```

NickStallman, 2024-11-13:
What HA version are you running by any chance?

Your HA is wanting power factor as a percentage which caused some discussion in https://github.com/home-assistant/architecture/issues/469 and https://github.com/home-assistant/core/pull/85181 since strictly speaking power factor is not a percentage but this should have been fixed last year.

I also don't see that message in my logs running on 2024.11.1.

mtrekker, 2024-11-14:
Comparing 2 sensors, one has and the other doesn't have unit defined.

[image]

NickStallman, 2024-11-29:
Yeah the problem is no unit is the correct one. :)
Power factor isn't actually supposed to be a percentage.

I've released version 0.2.2 which sets the unit of measurement to a space which appears to be the correct solution.
Can you re-check and see if this has done the trick?

closed 2025-01-09 by NickStallman