NickStallman/home-assistant-repo #35: Template variable warning: 'dict object' has no attribute 'value' when rendering '{{ value_json.value }}'.

pguest80, 2024-11-24:
Is it possible that the MQTT broker is causing this warning in Home Assistant?

Logger: homeassistant.helpers.template
Source: helpers/template.py:2748
First occurred: 18:18:02 (486 occurrences)
Last logged: 19:43:06

Warning: Template variable warning: 'dict object' has no attribute 'value' when rendering '{{ value_json.value }}'.

The sensor sensor.sg10rt_axxxxxxxxxxxxx_mppt_total_power is unknown when there is no sunlight.
Could this be the cause?

How can I avoid this if it is, or can I safely ignore it? It generates several thousand warnings per day.

Leahcim70, 2024-11-28:
@pguest80 is it possible that you just try to access a wrong value in a template?

NickStallman, 2024-11-29:
Yep this addon doesn't use templates, so it's something you've got referencing the Winet values.

closed 2024-11-29 by NickStallman

MeierAlex, 2025-02-26:
Hallo,

I have exactely the same problem like described in the initial post. I use some entities for my energy monitoring dashboard - cpuld this be the cause for the errors ?

Leahcim70, 2025-02-26:
> Hallo,
> 
> I have exactely the same problem like described in the initial post. I use some entities for my energy monitoring dashboard - could this be the cause for the errors ?

You are using a non existent entity in your dashboard.