NickStallman/home-assistant-repo #43: Additional documentation

brendan-pike, 2025-01-09:
Hi, sorry for opening a ticket I wasn't sure how to contact you.

I'm a Home Assistant newbie,  I'm sure for people with experience this is basic but I wanted to reach out and describe my trouble.   I've just got a Sungrow SH10RS setup and will be moving to Amber provider.  When looking to setup https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant I was advised somewhere that your winet is more reliable.  
So I've now setup HA Core with winet / mqqt and can see in the logs its detecting my inverter and battery with the sensors fine.

But at this point I'm stuck, it seems to me that to get good control automation I should use mkaiser's setup but I have no idea how to/if its possible integrate both. 

Can you please point me in the right direction?

NickStallman, 2025-01-09:
Yep, so the mechanism this addon uses is read only so it can get all the sensor values very quickly and reliably but it can't be used for controlling the inverter directly.

I use mkaiser for control, and just comment out sections of it that get duplicate data. Modbus does have some values this addon doesn't get like "inverter mode".
Relying on Modbus too heavily can lead to the dongle crashing more often, but just fetching a couple of things plus this addon using websockets for the frequent updates seems to work fine.

brendan-pike, 2025-01-09:
Ah thank you, that explains a lot which isn't apparent without experience.  Any chance you could share your config files to give me a jump start here?
I'd be happy to try and contribute back a howto for newbies like myself.

NickStallman, 2026-01-12:
This addon and mkaiser serve different purposes and work well together - no conflict at all.

This addon uses WebSocket to efficiently stream realtime sensor data (power, voltage, energy totals, etc). It's read-only but very reliable for frequent updates.

Mkaiser uses Modbus which is better for settings and control (changing modes, SOC limits, etc) and has some values this addon doesn't get.

I run both side by side - just disable the duplicate sensors in mkaiser's config to avoid overlap. The combination gives you the best of both worlds. :)