NickStallman/home-assistant-repo #44: Sensor Data Not Being Read

BB-Rabbit, 2025-01-12:
Hi Nick, Thank you very much for making this add-on. I found it very easy to add to my Home Assistant and read data from my Sungrow inverter.

Upon reviewing the data, it seems like not all of the data relating to the amount of power being exported to the grid is being shown. 

See below pic, is it possible for your add-on to grab the household usage data or power being exported to the grid? (2.4kW or 2.3kW respectfully). I am able to see the total solar generated, 4.7kW.

My setup has a Sungrow inverter: SG5.0RS and smart meter: SG Smart Meter. I assume the data from the smart meter is being fed into the sungrow inverter and that is how the data is populated in the pic below, but could be wrong.

Thanks in advance for any help.

[image]

NickStallman, 2025-01-13:
On the SH10RS (it may vary slightly) the export power is total_export_active_power and the household usage is total_load_active_power.

There's a whole bunch of sensors and some of the names are unintuitive but all the data should be there. :)

BB-Rabbit, 2025-01-13:
Thanks for the reply Nick,

Unfortunately i don't seem to have any sensors that resemble total_export_active_power or total_load_active_power.

See below screenshot of all the data that is being received.

Do you think I haven't configured the addon correctly?

[image]

rpassis, 2025-01-18:
@BB-Rabbit I have an SG80RS and am currently seeing the same as you. If I login to the WiNet interface I can see the exact same parameters and nothing else (they are even in the same order as your screenshot above). The weird thing is that the data in iSolarCloud is still showing production/import/export, I just don't understand where this data is actually coming from... 🤔 

[image]

[image]

rpassis, 2025-01-18:
I have a feeling this might be related to a firmware update and perhaps we are only seeing the inverter sensors and not longer seeing the smart meter ones?

BB-Rabbit, 2025-01-19:
Yes I agree with you @rpassis, I have a feeling a newer or older firmware/software version running in our Sungrow inverter is what is causing the Nick's addon to not grab all the data values.

I am not exactly sure how Nick's addon works, I assume maybe making API calls over the web interface?

@NickStallman would you be willing to give us some insight to how your addon grabs the inverter data? and then maybe I/we could do some troubleshooting on our systems to see if we can get to the bottom of it.

Thanks in advance.

BB-Rabbit, 2025-01-19:
@rpassis what firmware are you running on your WiNet-S dongle? mine is (M_WiNet-S_V01_V01_A)

omiks3, 2026-01-09:
I am having this exact issue too with a SG5.0RS inverter using a S100 meter. The data from the meter must be passed through the Winet as it is showing up in iSolarCloud per below. However when logging into the Winet it will only report meter phase A, B, C.

@BB-Rabbit have you made any progress with this? Btw I am running the same firmware on the Winet as you. I found that it is possible to get the Winet dongle to export all data in hex via https://192.168.20.7/#/system/dataSetExport/dataSetExport but have no idea how to interpret this. Sample of the data below:

2026-01-09 14:31:12.244 send data:01 65 03 00 02 1f 96 6f ab
2026-01-09 14:31:12.285 recv data:01 65 03 00 06 1f 96 00 00 00 00 29 34
2026-01-09 14:31:12.335 send data:01 65 03 00 02 1f 97 ae 6b
2026-01-09 14:31:12.384 recv data:01 65 03 00 0b 1f 97 00 00 00 01 16 68 01 13 88 e7 04
2026-01-09 14:31:12.446 send data:01 65 03 00 02 1f 98 ee 6f
2026-01-09 14:31:12.488 recv data:01 65 03 00 06 1f 98 00 00 00 00 40 f5
2026-01-09 14:31:12.552 send data:01 65 03 00 02 1f 99 2f af
2026-01-09 14:31:12.593 recv data:01 65 03 00 06 1f 99 00 00 00 00 7d 35
2026-01-09 14:31:12.657 send data:01 65 03 00 02 1f 9a 6f ae
2026-01-09 14:31:12.711 recv data:01 65 03 00 0f 1f 9a 00 00 00 01 14 d4 03 0c 8b 00 1e 0a a0 68 4b
2026-01-09 14:31:12.774 send data:01 65 03 00 02 1f 9b ae 6e
2026-01-09 14:31:12.830 recv data:01 65 03 00 14 1f 9b 00 00 00 02 14 5f 02 fd ff ff ff 15 e0 02 fd ff ff ff ef 2f

[image]

BB-Rabbit, 2026-01-09:
Hi @omiks3 sadly I have not made any more progress. I got busy with other projects and haven't cirlcued back to this, I am sure you know how it is!!!

Since my initial post, I have had some hardware changes to my solar system. I now have a battery and also have a larger inverter. 

So I now have a hybrid 8kW inverter: SH8.0RS. Battery Module: SBR128. Winet dongle: WiNet-S2.

See below the latest screen shots of what data is now showing. There seems to be a bucket load more data being outputted from the newer SH8.0RS inverter. or maybe it is the version 2 of the winet dongle? who knows.

Let me have a look at it today as the sun gets up and I can let you know.

[image]

[image]

[image]

closed 2026-01-12 by NickStallman