NickStallman/home-assistant-repo #45: Invalid Realtime message

veichnergmxde, 2025-01-15:
Hello Nick,

I have some Trouble at setting Up your AddOn.
The Mosquitto Broker ist running aswell AS the WiNet Extractor. In the log it seems that the connection to WiNet (Version 1) can be established and also the components like Wallbox, battery etc are discoverd. But no Itentities can be loaded. 
Do you have any Idea what the Problem might be?
The Converter ist SH8.0RT; i tried WiFi and Lan, both geht the Same Error.

`successfully started
> cli
> node build/src/index.js
2025-01-15 15:01:33 info: Fetched i18n properties. 
2025-01-15 15:01:34 info: Connected to MQTT broker 
2025-01-15 15:01:35 info: Connected to a older Winet-S device 
2025-01-15 15:01:35 info: Connected to Winet, logging in 
2025-01-15 15:01:35 info: Authenticated successfully 
2025-01-15 15:01:35 info: Detected device: SH80RT (A2282061072) 
2025-01-15 15:01:35 info: Detected device: SBR096 (S2211100369) 
2025-01-15 15:01:35 info: Detected device: AC011E01 (A2292804460) 
2025-01-15 15:01:37 error: Invalid realtime message: {
  data: {
    result_code: 1,
    result_msg: 'success',
    result_data: { service: 'real', list: [Array], count: 19 }
  },
  [Symbol(level)]: 'error',
  [Symbol(splat)]: [ { data: [Object] } ]
}
2025-01-15 15:01:37 warn: Reconnecting to Winet `

Vinc-boop, 2025-01-27:
Meanwhile i updated the Winet Firmware but the error keeps the same.
Would really look forward to get some advice from you.

`2025-01-27 20:18:49 info: Connected to MQTT broker 
2025-01-27 20:18:52 info: Fetched i18n properties. 
2025-01-27 20:18:55 info: Connected to a Winet-S2 device with newer firmware 
2025-01-27 20:18:55 info: Connected to Winet, logging in 
2025-01-27 20:18:56 info: Authenticated successfully 
2025-01-27 20:18:56 info: Detected device: SH80RT (A2282061072) 
2025-01-27 20:18:56 info: Detected device: SBR096 (S2211100369) 
2025-01-27 20:18:56 info: Detected device: AC011E01 (A2292804460) 
2025-01-27 20:18:57 error: Invalid realtime message: {
  data: {
    result_code: 1,
    result_msg: 'success',
    result_data: { service: 'real', list: [Array], count: 19 }
  },
  [Symbol(level)]: 'error',
  [Symbol(splat)]: [ { data: [Object] } ]
}
2025-01-27 20:18:57 warn: Reconnecting to Winet `

ulrichc580, 2025-02-03:
Hello @NickStallman , @Vinc-boop ,

@Vinc-boop you are not alone :-) 
I have same issue at my end. Connected devices (wallbox, battery and hybrid inverter) are discovered but the WiNet Extractor seem to have an issue reading or interpreting the data.

`2025-02-03 11:50:31 info: Connected to a older Winet-S device 
2025-02-03 11:50:31 info: Connected to Winet, logging in 
2025-02-03 11:50:31 info: Authenticated successfully 
2025-02-03 11:50:31 info: Detected device: SH10RT (A2341100907) 
2025-02-03 11:50:31 info: Detected device: SBR096 (S2209140137) 
2025-02-03 11:50:31 info: Detected device: AC011E01 (A2270800339) 
2025-02-03 11:50:32 error: Invalid realtime message: {
  data: {
    result_code: 1,
    result_msg: 'success',
    result_data: { service: 'real', list: [Array], count: 19 }
  },
  [Symbol(level)]: 'error',
  [Symbol(splat)]: [ { data: [Object] } ]
}
2025-02-03 11:50:32 warn: Reconnecting to Winet `

@NickStallman  would be great if you could somehow comment on that issue please. I would really appreciate using the WiNet Extractor to show the data in Homeassistant