NickStallman/home-assistant-repo #50: Winet add-on keeps stopping, sometimes within minutes of starting

dave373, 2025-02-08:
The addon will just die, continuously.  The longest I have seen it run in the last week is about 5 hours.

Usually, The only hint in the logs is "Killed". I have just caught a 400 error with a Node traceback, which I have not seen before.. maybe this can help?

It appears that it will try to restart a few times before giving up completely.  Then I have to go into the settings and start the add-on manually.

The fault started on the 31st January, as there is no missing data that I can see until then.

I have attached a log file that includes the traceback.
Also attached, A graph from 25th January - 8th February (internal temp) to show the drop outs from the add-on dying.

[image]

[winet-extractor_2025-02-08T11-48-43.055Z.log](https://github.com/user-attachments/files/18718518/winet-extractor_2025-02-08T11-48-43.055Z.log)

Inverter: SH10RS  
Winet Dongle:  Winet-S2,  Version M_WiNet-S2_V01_V01_A

icemirr0r, 2025-03-31:
Same here: Sometimes it works for a few hours/days and then it stopps. Only full reboot of HA fixes it.
Can happly provide logs etc.

dave373, 2025-03-31:
I am now running the add-on on a different server (non HA) and it doesn't die.  I suspect there is an issue with the supervisor killing it when it shouldn't.. I just can't find a way to prove this, apart from running it separately.