NickStallman/home-assistant-repo #56: User account seems to be locked

Ashleyde42, 2025-05-09:
I have been trying to connect to this for six months and cannot get past this error message.  I have updated all the firmware, changed passwords etc multiple times, but keep getting this user account lock.  Really want to get this working in home assistant, if you could offer a suggestion on how to resolve, it would really be appreciated.

> cli
> node build/src/index.js
2025-05-09 05:55:36 info: Connected to MQTT broker 
2025-05-09 05:55:37 info: Fetched i18n properties. 
2025-05-09 05:55:38 info: Connected to a Winet-S2 device with newer firmware 
2025-05-09 05:55:38 info: Connected to Winet, logging in 
2025-05-09 05:55:39 error: Invalid login message: {
  data: {
    result_code: 114,
    result_msg: 'I18N_COMMON_USR_ACCOUNT_LOCK',
    result_data: { service: 'login', errorcnt: 0, unlocktime: 9 }
  },
  [Symbol(level)]: 'error',
  [Symbol(splat)]: [ { data: [Object] } ]
}
/usr/src/app/build/src/winetHandler.js:400

mikecouchman, 2025-05-19:
I've just had a look at my configuration screen (Settings -> Add-ons -> WiNet Extractor then click on the configuration tab at the top, between Info and Log). I have my IP address as the winet_host, and my mqtt_url connection string, but I have nothing in both winet_user and winet_pass. Have you tried no username/password?

Ashleyde42, 2025-05-20:
I tried it deleting my Winet Userid and password.  That did it.  Well done!! thank you very much.
I have been trying this for six months and numerous HA support sites and Facebook but no one helped or could help.

Now I just have to work out how to get it out onto a dashboard.  Shame after so many months of trying and giving up I can't remember a lot of it.  But now I have the data with access to 78 sensors I am sure I will be able to figure it out.  I hope.

Thanks again.  It never occurred to me to drop the userid password.  I kept changing them to try and make certain they were all the same.

But thanks again.
Ash

mikecouchman, 2025-11-12:
@jda90000 Looks like your MQTT user/pass is incorrect. The connection string (under mqtt_url in WiNet Extractor Configuration) should look like mqtt://{mqttuseraccount}:{mqttpassword}@core-mosquitto:1883 - replace the {mqttuseraccount} and {mqttpassword} with your user and pwd you have setup.

In the WiNet Extractor Configuration screen, its the winet_user and winet_pass that should be left blank.

Also, turn on SSL.

jda90000, 2025-11-13:
Great, thank you Mike.
John

closed 2026-01-12 by NickStallman