NickStallman/home-assistant-repo #61: Sensors are not updated after upgrading to the latest HA (2025.7.2)

soundvibe, 2025-07-17:
Addon stopped updating data after I've updated my HA to the `2025.7.2` version. 
There's no errors in the logs, it prints e.g. `2025-07-17 08:05:09 info: Updated 24 sensors` but the sensors in the HA aren't updated.

wangeris, 2025-07-17:
might be related, getting this warning in 2025.6.3

`2025-07-17 11:21:02.065 WARNING (MainThread) [homeassistant.components.mqtt.sensor] The unit of measurement ` ` is not valid together with device class `power_factor`. this will stop working in HA Core 2025.7.0`

soundvibe, 2025-07-17:
> might be related, getting this warning in 2025.6.3
> 
> > 2025-07-17 11:21:02.065 WARNING (MainThread) [homeassistant.components.mqtt.sensor] The unit of measurement ` ` is not valid together with device class `power_factor`. this will stop working in HA Core 2025.7.0

In my case, it worked fine with HA 2025.7.1, but is stopped working right after upgrade to 2025.7.2

soundvibe, 2025-07-18:
It started to work again today, not sure what was causing this, closing the issue.